thomasonw/ATmegaxxM1-C1 #3: Board clock mismatch

scozzarro, 2017-11-23:
Dear thomasonw I'm encountering some problem with the clock of the board using your code. 
I noticed it because the serial comunication wich I set to 9600 instead is actualy at 600.
My board has a 16MHz xtal and I'm using the right bootlader version ATmega64M1 (16MHz - xtal - Arduino). Can you tell me more about it ? 
Many thanks.

thomasonw, 2017-11-24:
@scozzarro : Interesting, I have been using the Serial port with no problems, though it is behind an FT232 chip which could be masking things.  In a few days I will take a scope to the actual Serial signal and confirm the baud rates.  FWIW, I am using 115,200, and 57,500 is the rate used by the bootloader - that though is done via a soft serial, not using the hardware UART.

I am assuming you are using the bootloader as well as the Arduino libs from this repository, there was a LOT or work around the Serial port to get it to work - and work reliably.

Will let you know what I find and we can see if we can get to the bottom of the issue you are having.

thomasonw, 2017-11-27:
@scozzarro  Hello.  Just not I looked at Tx serial transmissions from the ATMega64M1 running at '115,200' baud, and the frequency looks correct to me.  (I measured 8.520uS per bit).   I am wondering what is happening in your case!

closed 2018-02-18 by thomasonw