thomasonw/ATmegaxxM1-C1 #4: ATmega16u2 USB-to-serial support

Frez-g, 2018-01-23:
Dear thomasonw, first of all thanks for your great work: it has been of extreme help in one of my recent project!
I don't know if this is the right place to open this issue but otherwise I wouldn't know who to ask to.
I'm trying to build a custom Arduino-style board developed around the ATmega32M1 which I managed to program with your bootloader effortlessy: to be able to use it exactly as an Arduino I've integrated in my pcb also a ATmega16u2 with the exact same connections used in an Arduino (TXD, RXD, CTS) to be able to program the 32M1 via USB taking advantage of the USB-to-serial firmware available for the 16u2. My problem is that although I'm able to load this firmware without any problem (using a USBasp programmer and avrdude) I cannot make the 16u2 communicate with the 32M1: when I try to upload a sketch in the same way that you would do it for an Arduino I receive the `avrdude device not in synch resp=0x00` error that usually occurs when you try to program the wrong Arduino board. Are you aware of any change that has to be made to that firmware to make the 2 chips talk (e.g. Product ID, communication frequency...)
If you feel like this is the wrong place to ask can you suggest me where to find an answer?

thomasonw, 2018-01-23:
Hello.

Glad you are finding this porting effort of value.  In my designs I have been using TTL to USB conversion chips, specifically the FT230XS-R, though there should be no reason a 16u2 device should not work.   

Question:  Once your board is programmed are you able to communicate via the Serial port / 16u2 / USB connection?  Meaning, are you able to proof that this path will work, and that only the bootloader is having issues?

In terms of the bootloader, a couple of points:  It is implemented using software bit-banding UART (as opposed to the hardware), and it is set to operate at 57,600 baud.  I do not think the software UART would cause any issues (as long as you have matched the xtal frequency with the bootloader flashed in) - but you might want to confirm the baud rate.

I will also add that I myself have never tested the 32M1 uC, only the 64M1 - it could also be that there is an issue there..

While debugging this port I ended up needing to use a O'Scope to watch the initial hand-shake during flashing.

Not sure if any of this helps - perhaps verify the baud rate, as well as verify if indeed you are able to communicate via serial.print() functions once programmed....

let me know how it goes and will see what can be done to help!

Frez-g, 2018-01-23:
Thank you for the informations.

Once the 16u2 is programmed it's recognized by my computer so I think I should be able to communicate with it. The problem is that it is seen as an Arduino Uno board because the firmware I use is the one used in the Uno: I don't know if the reason the system is not working is because the Arduino IDE is expecting to program an Arduino Uno but a 32M1 device is selected instead (even though I select ATmega32M1 with the correct crystal frequency as the board to be programmed).

thomasonw, 2018-01-23:
"....it is seen as an Arduino Uno board .."

Perhaps that is a clue.   You will need to select the ATmega32M1 board type to be able to program it, You might also try enabling the Verbose option for uploading (file/preferences in the Arduino IDE) - it might give a bit more of a clue to what is happening.

I have also used this DOS command to just 'ping' a board via avrdude:

`avrdude -C../etc/avrdude.conf -v  -patmega64m1 -carduino -PCOM14 -b57600 -e `

Of course you may have to change the paths and such to make it work (along with the COM port).

FWIW:  When I did my project I looked at using the 16u2 chip, but was able to source the FTTI part for less cost - and less work in the end : - )

You may well be onto something with the auto ID.   This ATmegaxxM1 port does not have an 'official' Arduino ID associated with it, and it could well be the default 16u2 programming is causing the issue by auto-selecting the Uno board??

thomasonw, 2018-02-18:
Closing, not sure if issue is still happening, and also noting CPU is the U1  not the M1 series.

closed 2018-02-18 by thomasonw