thomasonw/ATmegaxxM1-C1 #5: 32m1 Compatibility

geraldjust, 2018-01-26:
Hi, Love the support you brought these uC. But ive been having some issues with the 32m1 uC. 

i get the following error trying to burn the bootloader:

```
avrdude: verification error, first mismatch at byte 0x7e00
         0x00 != 0x11
avrdude: verification error; content mismatch
Error while burning bootloader.
```
Ive used the 64m1 just fine. ive ran the 64m1 with the int osc. then a external 8mhz and a 16mhz and all those seem fine. But i keep getting this error for the 32m1. Not sure if there is something wrong with the verification for the bits or a fuse is not burn the right way.

Thanks for all your hard work!

Gerald R.

thomasonw, 2018-01-27:
Hello,   and Thank you - am glad you are able to make use of this.

I have actually never tested the 32M1 variant, and suspect there is some issue in the avrdude.conf file.    This is while loading thee boot loader, right?   Can you turn on Verbose mode in Arduino, do a load attempt and then copy and paste the entire output window here?   Maybe we can see something from that.

Thanks,
-al-

geraldjust, 2018-02-09:
Sure thing! Here is what i get:

`avrdude: Version 6.0.1, compiled on Apr 14 2015 at 16:30:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/avrdude.conf"
         User configuration file is "/Users/geraldrubalcava/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 020:011
         AVR Part                      : ATMEGA32M1
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20    64    0 no       1024    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9584
avrdude: erasing chip
avrdude: Using SCK period of 10 usec
avrdude: reading input file "0x3F"
avrdude: writing lock (1 bytes):

/Users/geraldrubalcava/Library/Arduino15/packages/arduino/tools/avrdude/6.0.1-arduino5/bin/avrdude -C/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/avrdude.conf -v -patmega32m1 -cusbtiny -Uflash:w:/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/bootloaders/optiboot/optiboot_atmega32m1_16Mhz.hex:i -Ulock:w:0x0F:m 
Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3F:
avrdude: load data lock data from input file 0x3F:
avrdude: input file 0x3F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified
[... 98 lines cut ...]
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: Using SCK period of 10 usec
avrdude: reading input file "/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/bootloaders/optiboot/optiboot_atmega32m1_16Mhz.hex"
avrdude: writing flash (32768 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against /Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/bootloaders/optiboot/optiboot_atmega32m1_16Mhz.hex:
avrdude: load data flash data from input file /Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/bootloaders/optiboot/optiboot_atmega32m1_16Mhz.hex:
avrdude: input file /Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/bootloaders/optiboot/optiboot_atmega32m1_16Mhz.hex contains 32768 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x7e00
         0x00 != 0x11
avrdude: verification error; content mismatch

avrdude done.  Thank you.

Error while burning bootloader.`

geraldjust, 2018-02-09:
One thing i think im noticing is that its uploading Twice? not sure if that might be the issue.

thomasonw, 2018-02-09:
Twice, does look odd.   Seems to be doing a sample 1st, trying to change fuses / flash in a few bytes checking to make sure it worked (which is looks like it did)

Just looking over it I notice you are using an USBtinyISP, I have always used one of the USBasp clones from China (with re flashed firmware as needed, ref: [http://www.rogerclark.net/updating-firmware-on-usbasp-bought-from-ebay/](http://www.rogerclark.net/updating-firmware-on-usbasp-bought-from-ebay/) )

And I do not know what the line `avrdude: programmer operation not supported` means.

But you have flashed an ATMega64M1, so unlikely the USBtinyISP is not the issue.   

Overall nothing pops out here - other then the bootloader is not actually begin burned into Flash (ala, the Flash is still cleared and all set = 0) .   I do know I had issues at one time if the power supply not being sufficient to the board;  just relying on the programmers power was not enough and it caused issues.  I also have at items had to slow down the SPI interface while flashing, there is a jumper on the USBasp for this.  But most of these 'issues' was when I was working with a 3.3v / 8Mhz design.

OK - it will take me several days, but let me dig out some parts and solder up an ATMega32M1 of my own and see what I can make happen.  If any of the above helps you find a solution, great!  Else, give me some time to mock up a board here see what happens on my end...

geraldjust, 2018-02-09:
Well i had absolutely no problem with the 64m1. And Sure! if you need a PCB design i can send you one. I have tested the 64m1 with int8mhz, ext8mhz, and ext16mhz and they all work very well! And thank you! I wish i could help more.

geraldjust, 2018-02-10:
just to help as much as posible here is my verbose mode for the 64m1. Which is working just fine:

`/Users/geraldrubalcava/Library/Arduino15/packages/arduino/tools/avrdude/6.0.1-arduino5/bin/avrdude -C/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/avrdude.conf -v -patmega64m1 -cusbtiny -e -Ulock:w:0x3F:m -Uefuse:w:0x1D:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.0.1, compiled on Apr 14 2015 at 16:30:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/avrdude.conf"
         User configuration file is "/Users/geraldrubalcava/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 020:009
         AVR Part                      : ATMEGA64M1
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20    64    0 no       2048    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes     65536  256    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9684
avrdude: erasing chip
avrdude: Using SCK period of 10 usec
avrdude: reading input file "0x3F"
avrdude: writing lock (1 bytes):

/Users/geraldrubalcava/Library/Arduino15/packages/arduino/tools/avrdude/6.0.1-arduino5/bin/avrdude -C/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/avrdude.conf -v -patmega64m1 -cusbtiny -Uflash:w:/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/bootloaders/optiboot/optiboot_atmega64M1_16Mhz.hex:i -Ulock:w:0x0F:m 
Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3F:
avrdude: load data lock data from input file 0x3F:
avrdude: input file 0x3F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

[... 111 lines cut ...]
avrdude: load data flash data from input file /Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/bootloaders/optiboot/optiboot_atmega64M1_16Mhz.hex:
avrdude: input file /Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/bootloaders/optiboot/optiboot_atmega64M1_16Mhz.hex contains 65536 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 65536 bytes of flash verified
avrdude: reading input file "0x0F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0F:
avrdude: load data lock data from input file 0x0F:
avrdude: input file 0x0F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.`

thomasonw, 2018-02-18:
Hello.

Today I soldered up an ATmega32M1 demo board and was able to upload the bootloader without any issue.  I am using the USBasp device, and here is the capture of it loading.

```
/home/al/.arduino15/packages/arduino/tools/avrdude/6.0.1-arduino5/bin/avrdude -C/home/al/.arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.6/avrdude.conf -v -patmega32m1 -cusbasp -Pusb -e -Ulock:w:0x3F:m -Uefuse:w:0x1D:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.0.1, compiled on Apr 14 2015 at 19:04:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/home/al/.arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.6/avrdude.conf"
         User configuration file is "/home/al/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATMEGA32M1
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20    64    0 no       1024    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9584
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: reading input file "0x3F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3F:
avrdude: load data lock data from input file 0x3F:
avrdude: input file 0x3F contains 1 bytes
avrdude: reading on-chip lock data:

[... 121 lines cut ...]
avrdude: verifying ...
avrdude: 32768 bytes of flash verified
avrdude: reading input file "0x0F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0F:
avrdude: load data lock data from input file 0x0F:
avrdude: input file 0x0F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.

```

I was able to do this on a Windows machine, as well as Linux - however I did find an issue with Linux:   it was looking for the hex file with the name 32m1.hex, while the actual filename was 32M1.hex (case of the letter M).  Windows did not care, but Linux said the file was not located.  As a temp fix simple changed the name of the .hex file.

At this point I would suggest perhaps trying a different ISP device and see if still are having issues.  I will be looking into how to correct the filename letter M case for those running on Linux and the like.

thomasonw, 2018-02-26:
Closing this as issue.   I was able to Flash an ATmega32M1 using a USBasp with no noted errors (See above).  If problem still persists please reopen this issue.

closed 2018-02-26 by thomasonw