thomasonw/ATmegaxxM1-C1 #6: How to control DAC

Jiloc, 2018-01-29:
Hello, I have a problem with the DAC. When I try to write on the D10 with an analogWrite it seems to ignore it.

Example:
`
#define DAC 10

void setup() {
    pinMode(DAC, OUTPUT); 
}
void loop(){
    analogWrite(DAC, 512);
}
`

When I physically read the voltage with the multipler the voltage stays fixed at 0.21 volts. I also tried to change the value (ranging from 0 to 1023) but it doesn't change.

When I instead use `digitalWrite(DAC, HIGH)` it works. Any ideas?

thomasonw, 2018-02-18:
@Jiloc   Hello.  Sorry for the long delay.  This message is just to let you know I have a mockup board I can use to look into what you have noted above.  Should have more to say in a couple of days.

Jiloc, 2018-02-19:
Hello @thomasonw, 
do not worry for the delay, rather thanks for the reply and for making this project. 
Anyway it seems like my board had some hardware problems with the SLEEP and RESET pins, so the only problem is with the DAC that doesnt work as an alogic pin as it should. I just can control it as digital

thomasonw, 2018-02-26:
Hello.

Today I was able to sit down and work on this issue.

1st, I was able to reproduce the DAC problem.  It was that the VRef was not set, and hence the DAC had no reference voltage.   If one adds this in the setup() section, your example will work:

`       ADMUX = DEFAULT << 6;              // Make sure the Vref is selected.'

However I was not able to reproduce your issues with the digital writes to A3 and A4.  It worked OK for me, testing on an ATmega64M1.   You example, I also tried the BLINK example, as well as modified the above example to a 'Blink' style toggling both ports.  I do know that A3 and A4 are two of the special PSC ports and held to zero until released.  There was a bug fixed in the lib a while ago, you will want to make sure you have the latest lib.

I have pushed this change up to Github, and will wait a little before creating a new version.  If you get a chance to test this would be appreciated.  As well, if you can confirm the A3/A4 issue -- perhaps sending some details of the hardware you are using?

Jiloc, 2018-02-26:
Hello, thank you for the fix! For what concerns A3 and A4 instead, as I reported in my last comment, it was just an hardware problem of my board, I am waiting for the replacement that I should i receive in a few days. When I'll have it I'll test the DAC fix and report the result here. Thank you again for the amazing work

closed 2018-04-24 by thomasonw