thomasonw/ATmegaxxM1-C1 #8: ATmega64m1-MZ

geraldjust, 2018-03-27:
By any chance do you know if there is a difference between the regular AU and the MZ version of the chip? I made a board with both types. They Both seem to work just fine when i try to burn the bootloader and the fuse bytes. but as soon as it does that. All the pins are pulled High. Checked and it has the same device signature and everything. The only thing digging around i found this 1611-ATMEGA64M1-15MZCT-ND digikey part number and the following pdf.

http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7647s-Automotive-Microcontrollers-ATmega16M1-32M1-64M1-32C1-64C1_Datasheet.pdf

Thank you!

thomasonw, 2018-03-27:
Yes,  it defies the package:  TQFP vs. leadless QFN.

 

Ref the ‘Ordering Information’ section in the spec sheet for details.

 

-al-

 

http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7647s-Automotive-Microcontrollers-ATmega16M1-32M1-64M1-32C1-64C1_Datasheet.pdf

geraldjust, 2018-03-27:
Oh Yes! i know that. And the cost between the two is vastly different. One cost 1.20 and the other cost 6 dollar. But i made a breakout board for both. They both have the same device signature. but when i burn the bootloader Everything says it goes well. But then all the pins go Into a high State. And i cant seems to get the chip do do anything. not even uploading though serial. So not sure if there was any difference in some registers. Or something like that.

Thanks!

thomasonw, 2018-03-27:
Oh, sorry.  Hum...  can not for the life of me see any electrical differences between them..  No reason why one should work and the other not, unless there was an assembly issue (soldering) with the lead-less package?  Will be VERY interested in hearing the resolution to this, but sorry can not think of anything at the moment which might help explain what you are seeing - at least from a software viewpoint.

geraldjust, 2018-03-27:
I don't think there is any electrical differences between the two. But from what i can tell, there was some updates on this package mid 2017.  Microchip remaned it to the ATmega64m1-Automotive. The only thing i can see electrically is some minor power consumption stuff, and the operating temp is much higher. I asked for new samples of there chips. See if there the same as the ones on Digikey.

If you need some boards too let me know! im more than happy to ship you my designs! Just want to debug this thing.

Il start off by looking at the PSC and Some other fuse bits see if anything is creating this problem.

geraldjust, 2018-03-28:
The Strange thing! i put together a fresh chip, And the factory fuses are:
L:0x62
H:0xDF
E:0xFF
Lock:0x3F
And even like this. All pins are set to high. But they all output 3.5V (VCC is 5v),  Set a regular ISP programmer is able to communicate and i get the same device signature. When the Reset Is set low. All pins then float. 

When reading fuse bits or programming. The signal levels actually go from 0-5v. Then right after it goes back to 3.5v ish. 

Im going to dig a bit deeper but.. perhaps Thats why digikey has these chips at $1.12 ? lol

thomasonw, 2018-03-29:
Hum...   Perhaps.  Glad you started this thread, as I was looking into making a batch build using those CPUs!  Will be interested in hearing the conclusion.

geraldjust, 2018-03-29:
im going to try ATmega64m1-15MD part number. See what's different. Its still in the smaller package but the price is around $1.50. From what all the datasheets i've been reading into, seems that this is part of the older markings. So there might be a chance this is closer to the AU part number.

I Went though Microchips website. And my suspicions were correct about the MZ. In there system its considered a new part along with the AZ. Its automotive grade now. Which im not sure why.. if the 64c1 main reason was that it was just that, for (auto) use. Il get back to ya to see if the MD works like the AU line does.

geraldjust, 2018-03-30:
Would there be a problem if i left Aref floating? im trying to go though my hardware designs to double check. but I cant seems to find anything wrong except for that. Tried the MD,  and i got the same thing.  when in ISP mode when uploading all pins go low. Then once done it floats around 2.0-3.7ish.  I tried uploading and doing pin manipulations. and i did not get anything done so far.

thomasonw, 2018-03-31:
You shouldn’t NEED anything on AREF, at least not at this point in your testing.   Is very odd you are able to talk to the CPU (via the ISCP I am assuming), but not getting much response.

 

Oh!  If you are trying to wiggle the ports directly (vs using Arduino libs) one point, the fuses I have selected will disable any I/O pins used by the PSC subsystem at POR until they are released.  Ref “25.3 PSC Output Behavior during Reset” in the datasheet.

geraldjust, 2018-04-02:
YES! That is what i thought was so strange! when the reset line is pulled, and in ISP mode, the pins swing from 0-5v for the data, and i can see it. But as soon as that's done, they start floating im guessing. 

and i was trying to control the pins with standard arduino rather then Atmel studio. And those PSC fuses pull the pins low until there used by the main program right?

have you used any of the QFN packages yourself?

thomasonw, 2018-04-02:
Hello.

So, does it look like this explains things? Do you have the QFN package device working well now?

If you are using my Arduino board libs here, I have put special code in to the standard pinMode(), writes, etc.  calls to enable the pins after reset.  You should be able to load up Blink.ino and run without any issues, are you able to?

No, at present I have not used the QFN packages.

geraldjust, 2018-04-02:
Yea i was guessing that. But no i haven't gotten it to work. I will give it another look soon. But here is a more detailed explanation.  Im able to  use ISP to "burn" the bootloader. But even though its burnt in there. i still cant use the serial to upload new sketches. And all pins before, and after, seem to be floating. I also tired uploading a blink program straight though the ISP. And i get weird result. I set only one pin to LOW, and i check all the others and they all seem to go haywire. Some stay floating. Others two or three pins go LOW... So not sure if there was any revision in these chips...BUT

I just ordered a third set of chips this time its the ATMEGA64M1-MU which is from the same line as the AU. Just wanted to rule out by breakout board design. But these chips are 5>$ vs the MD&MZ that are 1.50<$

thomasonw, 2018-04-02:
Hum…   Almost sounds like a bad / missing ground pin somewhere????

 

Very interested in what you figure out!

geraldjust, 2018-04-03:
I thought so too. But in that case i would not be able to be programmed. So if This next chip does work , with my board design then i can conclude there might be something different with the other two chips that im not aware of. And if it also doesnt work, then il go back to the drawing board again. Or see if Atmel Studio could work with the other model of the MD & MZ part numbers.

geraldjust, 2018-04-03:
the center pad in my design i connected it to the ground plane. but i took a fresh chip out and it doesnt seem to be connected to ground or vcc or any pin for that matter. Other then that... not sure..

thomasonw, 2018-04-15:
Hey, wonderful!  yes, those leadless packages can be a real issue to get soldered correctly – esp by hand.

 

Thanks for the feedback!

thomasonw, 2018-04-15:
Hum…  That is a bit disturbing that you had solder issues using the equipment you noted…

 

-al-

geraldjust, 2018-04-24:
Confirmed! longer pads and i added a xap from Aref to Ground. And the QFN-MZ is working well.

Gerald R.

thomasonw, 2018-04-24:
Hey, that is great news!  Thanks for following up.

closed 2018-04-24 by thomasonw