thomasonw/ATmegaxxM1-C1 #9: boards.txt updating

geraldjust, 2018-04-25:
Hey! sorry to bug again. I have been working on a updated board.txt file. So i have made it compatible with the file name you have. Its been compiling just fine. I even added a menu on the board option for the PSC to be OFF, LOW, HIGH. 

But when uploading it, ive noticed that its using the arduino avrdude.conf rather then the custom one in the core hardware folder. so i get a error saying the 64m1 is not available.

Here is the new code for the boards.txt:

`############################################################

AtmegaX1.name=ATmegaX1
AtmegaX1.upload.tool=arduino:avrdude
AtmegaX1.upload.protocol=arduino
AtmegaX1.build.variant=ATmegaxxM1

AtmegaX1.bootloader.tool=arduino:avrdude
AtmegaX1.bootloader.unlock_bits=0x3F
AtmegaX1.bootloader.lock_bits=0x0F

AtmegaX1.menu.variant.64m1=ATmega64m1
AtmegaX1.menu.variant.64m1.build.mcu=atmega64m1
AtmegaX1.menu.variant.64m1.build.board=ATmega64M1
AtmegaX1.menu.variant.64m1.build.name2=atmega64M1
AtmegaX1.menu.variant.64m1.upload.maximum_size=64512
AtmegaX1.menu.variant.64m1.upload.maximum_data_size=4096

AtmegaX1.menu.variant.32m1=ATmega32m1
AtmegaX1.menu.variant.32m1.build.mcu=atmega32m1
AtmegaX1.menu.variant.32m1.build.board=ATmega32M1_16xtl
AtmegaX1.menu.variant.32m1.build.name2=atmega32M1
AtmegaX1.menu.variant.32m1.upload.maximum_size=32256
AtmegaX1.menu.variant.32m1.upload.maximum_data_size=2048

AtmegaX1.menu.PSC.OFF=OFF
AtmegaX1.menu.PSC.OFF.bootloader.extended_fuses=0x3D
AtmegaX1.menu.PSC.LOW=LOW
AtmegaX1.menu.PSC.LOW.bootloader.extended_fuses=0x1D
AtmegaX1.menu.PSC.HGH=HIGH
AtmegaX1.menu.PSC.HIGH.bootloader.extended_fuses=0x05

##############################################################

#make atmegaXXm1 AVR_FREQ=16000000L BAUD_RATE=57600
AtmegaX1.menu.clock.16MHz_external.build.core=arduino
AtmegaX1.menu.clock.16MHz_external.build.f_cpu=16000000L
AtmegaX1.menu.clock.16MHz_external=16MHz External
AtmegaX1.menu.clock.16MHz_external.upload.speed=57600
AtmegaX1.menu.clock.16MHz_external.bootloader.low_fuses=0xFF
AtmegaX1.menu.clock.16MHz_external.bootloader.high_fuses=0xDE
AtmegaX1.menu.clock.16MHz_external.bootloader.file=optiboot/optiboot_{build.name2}_16Mhz.hex

#make atmegaXXm1 AVR_FREQ=16000000L BAUD_RATE=57600
AtmegaX1.menu.clock.8MHz_external.build.core=arduino
AtmegaX1.menu.clock.8MHz_external.build.f_cpu=8000000L
AtmegaX1.menu.clock.8MHz_external=8MHz External
AtmegaX1.menu.clock.8MHz_external.upload.speed=57600
AtmegaX1.menu.clock.8MHz_external.bootloader.low_fuses=0xFF
AtmegaX1.menu.clock.8MHz_external.bootloader.high_fuses=0xDE
AtmegaX1.menu.clock.8MHz_external.bootloader.file=optiboot/optiboot_{build.name2}_8Mhz.hex

#make atmegaXXm1 AVR_FREQ=8000000L BAUD_RATE=57600
AtmegaX1.menu.clock.8MHz_internal.build.core=arduino
AtmegaX1.menu.clock.8MHz_internal.build.f_cpu=8000000L
AtmegaX1.menu.clock.8MHz_internal=8MHz Internal
AtmegaX1.menu.clock.8MHz_internal.upload.speed=57600
AtmegaX1.menu.clock.8MHz_internal.bootloader.low_fuses=0xF2
AtmegaX1.menu.clock.8MHz_internal.bootloader.high_fuses=0xDE
AtmegaX1.menu.clock.8MHz_internal.bootloader.file=optiboot/optiboot_{build.name2}_8MHz.hex

############################################################`

do you know what links , or makes it use the avrdude.conf from the custom file?

thomasonw, 2018-04-25:
This looks nice!  Will be a good addition.

 

It has been a while from when I looked at this all, but I believe the answer is in the platform.txt file, and if not that, the programmers.txt  -- IIRC, there linkage is made via the .name, so if the name is changed from “ATmega64M1”  to say “ATmegaX1”, that could be the issue – but to be honest, it had been a few years that I played with it all…

geraldjust, 2018-04-25:
Being able to turn PSC off and on is a great feature i was looking for. Interesting thing is that uploading with ISP works just fine. Im able to change the fuse bits and PSC just fine with avrdude.conf and its compiling just fine too by adding "AtmegaX1.build.variant=ATmegaxxM1" which links it. The only issue is that once i upload via Serial, it defaults to the original arduino avrdude.conf

But thanks! if you find something let me know! But i will take a look at it too.

per1234, 2018-04-25:
I believe the reason for this is that you are referencing the Arduino AVR Boards avrdude tool:
```
AtmegaX1.upload.tool=arduino:avrdude
```
You'll note that in thomasonw's boards.txt it uses this packages avrdude tool:
```
atmega64m1_16Mhz_xtl.upload.tool=avrdude
```

geraldjust, 2018-04-26:
Perfect! I totally missed that! lol its working 100% now. 😃  Should be nice now to be able to turn on/off the PSC. I will add BOD perhaps too. 

Here is my updated boards.txt :
############################################################

menu.clock=Clock
menu.PSC=PSC
menu.variant=Variant

############################################################

AtmegaX1.name=ATmegaX1
AtmegaX1.upload.tool=avrdude
AtmegaX1.upload.protocol=arduino
AtmegaX1.build.variant=ATmegaxxM1
AtmegaX1.build.core=arduino

AtmegaX1.bootloader.tool=avrdude
AtmegaX1.bootloader.unlock_bits=0x3F
AtmegaX1.bootloader.lock_bits=0x0F

AtmegaX1.menu.variant.64m1=ATmega64m1
AtmegaX1.menu.variant.64m1.build.mcu=atmega64m1
AtmegaX1.menu.variant.64m1.build.board=ATmega64M1
AtmegaX1.menu.variant.64m1.build.name2=atmega64M1
AtmegaX1.menu.variant.64m1.upload.maximum_size=64512
AtmegaX1.menu.variant.64m1.upload.maximum_data_size=4096

AtmegaX1.menu.variant.32m1=ATmega32m1
AtmegaX1.menu.variant.32m1.build.mcu=atmega32m1
AtmegaX1.menu.variant.32m1.build.board=ATmega32M1_16xtl
AtmegaX1.menu.variant.32m1.build.name2=atmega32M1
AtmegaX1.menu.variant.32m1.upload.maximum_size=32256
AtmegaX1.menu.variant.32m1.upload.maximum_data_size=2048

AtmegaX1.menu.PSC.OFF=OFF
AtmegaX1.menu.PSC.OFF.bootloader.extended_fuses=0x3D
AtmegaX1.menu.PSC.LOW=LOW
AtmegaX1.menu.PSC.LOW.bootloader.extended_fuses=0x1D
AtmegaX1.menu.PSC.HGH=HIGH
AtmegaX1.menu.PSC.HIGH.bootloader.extended_fuses=0x05

##############################################################

#make atmegaXXm1 AVR_FREQ=16000000L BAUD_RATE=57600
AtmegaX1.menu.clock.16MHz_external.build.core=arduino
AtmegaX1.menu.clock.16MHz_external.build.f_cpu=16000000L
AtmegaX1.menu.clock.16MHz_external=16MHz External
AtmegaX1.menu.clock.16MHz_external.upload.speed=57600
AtmegaX1.menu.clock.16MHz_external.bootloader.low_fuses=0xFF
AtmegaX1.menu.clock.16MHz_external.bootloader.high_fuses=0xDE
AtmegaX1.menu.clock.16MHz_external.bootloader.file=optiboot/optiboot_{build.name2}_16Mhz.hex

#make atmegaXXm1 AVR_FREQ=16000000L BAUD_RATE=57600
AtmegaX1.menu.clock.8MHz_external.build.core=arduino
AtmegaX1.menu.clock.8MHz_external.build.f_cpu=8000000L
AtmegaX1.menu.clock.8MHz_external=8MHz External
AtmegaX1.menu.clock.8MHz_external.upload.speed=57600
AtmegaX1.menu.clock.8MHz_external.bootloader.low_fuses=0xFF
AtmegaX1.menu.clock.8MHz_external.bootloader.high_fuses=0xDE
AtmegaX1.menu.clock.8MHz_external.bootloader.file=optiboot/optiboot_{build.name2}_8Mhz.hex

#make atmegaXXm1 AVR_FREQ=8000000L BAUD_RATE=57600
AtmegaX1.menu.clock.8MHz_internal.build.core=arduino
AtmegaX1.menu.clock.8MHz_internal.build.f_cpu=8000000L
AtmegaX1.menu.clock.8MHz_internal=8MHz Internal
AtmegaX1.menu.clock.8MHz_internal.upload.speed=57600
AtmegaX1.menu.clock.8MHz_internal.bootloader.low_fuses=0xF2
AtmegaX1.menu.clock.8MHz_internal.bootloader.high_fuses=0xDE
AtmegaX1.menu.clock.8MHz_internal.bootloader.file=optiboot/optiboot_{build.name2}_8MHz.hex

############################################################

geraldjust, 2018-04-26:
Yea! i would love to make it into a pull Request. Once i finished with anything else i will add.

thomasonw, 2018-04-26:
Great, thanks -- that you you get full credit!

Will pull it in and make up a new lib version in the next week or so.  Thanks for this!

closed 2018-04-26 by thomasonw