thomasonw/ATmegaxxM1-C1 #11: Vcc reading

Alexivia, 2018-09-14:
Hi,
When I use analogRead(A9) to get the Vcc/4 value, it returns a value around 153 for a 5V supply. If I decrease the supply voltage, this value increases. I can't find any information in the datasheet on what reference voltage is used when the ADC selects the Vcc/4 channel either, so I'm clueless on what the returned value means.
Has anyone here used this feature before?

Thank you in advance!
Regards.

geraldjust, 2018-09-14:
I think A9 it for the internal Temp sensor. A10 should be Vcc/4.

Alexivia, 2018-09-14:
Hi @geraldjust 
You are correct, I swapped those unintentionally... Either way, do you know the scaling factor of both A9 and A10 readings?

thomasonw, 2018-09-14:
Hello.   the ATmega64M1 datasheet talks about the internal temp sensor, section 21.8 in my copy.  Seems one needs to select the internal 2.56v reference and also section 21.8.1 talks about the scaling of this sensor.

Vcc/4 is used to sense Vcc, and again needs to be done using the internal 2.56v reference - it would then allow you to 'measure' Vcc begin supplied to the CPU using 2.56v as the reference point.

Does this help?

Alexivia, 2018-09-17:
@thomasonw thank you for your answer. Applying the 2.56V reference to both A9 and A10 measurements, and using the mv/ºC scaling factor to the temperature sensor corrected my measurements. Thank you for your support.

Regards

closed 2018-09-17 by Alexivia

thomasonw, 2018-09-17:
Hey, wonderful!   Glad you got it working and verifying the functionality!