thomasonw/ATmegaxxM1-C1 #16: Turn off the timer0

alessio31183, 2018-12-19:
I've found an issue on using this platform.
If I turn off the timer0 with the instruction Turn off the TCCR0B = 0 nothing works.

Any suggestion?

thomasonw, 2018-12-19:
IIRC, TIMER-0 is used for the heart-beat, ala millis() , but also other internal timing related functions.   I suspect if you turn it off some of the Arduino CORE files will be caught in an infinite loop waiting for a timeout to occur.

alessio31183, 2018-12-19:
Yes I've found that the timer0 is always on for some arduino tick count.
Timer0 is set in the wiring.c inside the arduino core folder

thomasonw, 2018-12-20:
Ya, I think you will find Timer-0 is rather critical to the whole Arduino approach.  Any specific reason you are looking to turn it off?

alessio31183, 2018-12-20:
Because I had an idea that it turns on the uC sleep mode.

Anyway solved.

Thanks