thomasonw/ATmegaxxM1-C1 #18: Bootloader source code

MCUdude, 2019-04-21:
Hi!

Could you please upload the actual source code you used when you compiled Optiboot for these AVRs? I'd like to add ATmega16/32/64M1 and ATmega32C1/64C1 to my [Optiboot flash repo](https://github.com/MCUdude/optiboot_flash) and was hoping to use the source code (particularly the contents of pin_defs.h) as a template when adding these chips.

thomasonw, 2019-04-22:
Sounds like a great idea!   Sadly, I do believe I have lost the source I used for that build from several OS changes  (Sign, still miss WinXP).

I looked over my current computer and can not find anything.  There is a CHANCE it might be around on some old backups and /  old computers.   But that may well take a month or so before I am in a place where I can access them.  Will keep it in mind and try to look as soon as I can, because yes - it would be good to have the build files archived here...

MCUdude, 2019-04-22:
A little Google-fu reveals that there exists a [pull request](https://github.com/Optiboot/optiboot/pull/176) for adding ATmega16M1/32M1/64M1 support to the official Optiboot repo. I might try to copy it over to my own repo and add it to the [giant build script](https://github.com/MCUdude/optiboot_flash/blob/master/makeall). May I ask you if you have any hardware that may be used for testing? Since these microcontrollers differ so much from the rest of the AVR families it would be great if the bootloader could be tested on real hardware before making it public.