thomasonw/ATmegaxxM1-C1 #26: Install Compatibility with Arduino IDE 2.0

Slep-wt, 2024-02-14:
Hi there, just wondering if this package is compatible with Arduino IDE 2.0? When I try to install, I'm hit with:
```
Tool arduino:avr-gcc@4.8.1-arduino5 already installed
Tool arduino:avrdude@6.0.1-arduino5 already installed
Downloading packages
ATmegaM1-C1:avr@1.0.8
Installing platform ATmegaM1-C1:avr@1.0.8
Failed to install platform: 'ATmegaM1-C1:avr:1.0.8'.
Error: 13 INTERNAL: Cannot install platform: installing platform ATmegaM1-C1:avr@1.0.8: extracting archive: open C:\Users\Admin\AppData\Local\Arduino15\tmp\package-1226863038: is a directory
```
Thanks :)

E-J-Harris, 2024-05-14:
Hi Folks, 

I'm getting the same issue.  Is there a way to use this with Arduino IDE 2.3.2?

jblazeg, 2024-05-23:
Same problem here - anyone got a solution?